stm32-hotspot/stm32wb-matter-device-over-thread #1: NUCLEO-WB55RG Matter Thread end device commissioning failed

pavansrkote1, 2023-05-31:
Hello,

I am trying to commissioning NUCLEO-WB55RG Matter thread end device over thread but it fails to commission and connect.

Reference Source Code used:
Matter beta version for STM32 Github repo: https://github.com/stm32-hotspot/stm32wb-matter-device-over-thread  

Setup:

1. Matter Thread End Device

Below binaries flashed on the STM32 Matter Thread End Device(NUCLEO-WB55).

1.1. FUS: "stm32wb-matter-device-over-thread-main\Projects\STM32WB_Copro_Wireless_Binaries\STM32WB5x\stm32wb5x_FUS_fw.bin" 

1.2. BLE Stack: "stm32wb-matter-device-over-thread-main\Projects\STM32WB_Copro_Wireless_Binaries\STM32WB5x\stm32wb5x_BLE_Thread_ForMatter_fw.bin" 

1.3. Application: "stm32wb-matter-device-over-thread-main\Projects\P-NUCLEO-WB55.Nucleo\Applications\Matter\Lighting-App\Lighting-App.bin"

2. Border router 

Raspberrypi(Host) 

RCP Device(Silabs RCP device)

3. Host as a mobile app(chiptool)

Also find the Attached logs captured on End device and Android Application while doing commissioning.

Please share us the working logs if any & also share us the working procedure as well working source code or binaries.

 

Thank you

Pavan

STYoannZamaron, 2023-06-12:
Hello,
Can we close this non-issue?
Best Regards.

pavansrkote1, 2023-06-13:
Hello,

Yes we can close.

Best Regards,
Pavan

closed 2023-06-13 by STYoannZamaron

Gonta01, 2023-07-07:
> 1.1. FUS: "stm32wb-matter-device-over-thread-main\Projects\STM32WB_Copro_Wireless_Binaries\STM32WB5x\stm32wb5x_FUS_fw.bin"
>
> 1.2. BLE Stack: [...]

How to properly do it? Should CubeProgrammer be used? What addresses must be used?