stm32-hotspot/stm32wb-matter-device-over-thread #2: Official Link for "ForMatter" Coprocessor Binary

chrismerck, 2023-06-19:
This repository contains a binary called `stm32wb5x_BLE_Thread_ForMatter_fw.bin`. Release notes in this repo indicate a version number of 1.0.1. Presumably this is officially signed by ST, however it's being released here in an unofficial (presumably third-party) repository. Can someone explain what is going on here? (I want to build a commercial product on top of this IC, and the binary stack is a very important but mysterious aspect of this chip.)

STYoannZamaron, 2023-06-20:
Hello,
This binary is used only for this project and is not supported by ST for other applications. If you want to use the official version, you can download the one from the latest cube Firmware. 
thanks

STYoannZamaron, 2023-06-21:
Hello,
the static use ble or thread radio, but not at the "same time".  Each time you want to change from ble or thread you need to re init the ip. In the dynamic binary you can use ble and thread at the same time.  For example, for matter we used the dynamic one because we need ble and thread at the same time for the commissioning.

closed 2023-07-26 by STYoannZamaron