stm32-hotspot/stm32wb-matter-device-over-thread #4: Failed to connect to Google Nest Hub 2nd Generation

Gonta01, 2023-07-07:
Product ID (0xFFF1) and Vendor ID (0x8004) were added to Google Developer Console.
During Commissioning process Blue led was turned on then turned off. Next green led was turned on.
Logs:
[0m[M4 APPLICATION] Matter Over Thread Lighting-App starting...
[0m[M4 APPLICATION]  Init matter layer 
[0m[M4 APPLICATION] Init CHIP stack
CHIP:DL: BLEManagerImpl::Init() complete
[0m[M4 APPLICATION] 1- Initialisation of BLE Stack...
[0m[M4 APPLICATION] 2- Initialisation of OpenThread Stack. FW info :
[0m[M4 APPLICATION] **********************************************************
[0m[M4 APPLICATION] WIRELESS COPROCESSOR FW:
[0m[M4 APPLICATION] VERSION ID = 1.0.1
[0m[M4 APPLICATION] FW Type : Dynamic Concurrent Mode BLE/Thread for Matter 
[0m[M4 APPLICATION] **********************************************************
[0m[M4 APPLICATION] Start init matter
CHIP:-: Current Software Version: 1.0
CHIP:SVR: Server initializing...
CHIP:TS: Last Known Good Time: [unknown]
CHIP:TS: Setting Last Known Good Time to firmware build time 2023-07-07T13:09:56
CHIP:DMG: AccessControl: initializing
CHIP:DMG: Examples::AccessControlDelegate::Init
CHIP:DMG: AccessControl: setting
CHIP:DMG: DefaultAclStorage: initializing
CHIP:DMG: DefaultAclStorage: 0 entries loaded
CHIP:ZCL: Using ZAP configuration...
CHIP:DMG: AccessControlCluster: initializing
CHIP:ZCL: Initiating Admin Commissioning cluster.
CHIP:ZCL: On/Off set value: 1 0
CHIP:ZCL: On/off already set to new value
CHIP:ZCL: Cluster LevelControl: attribute CurrentLevel set to hu
CHIP:-: LightMgr:LEVEL: lev:64->1
CHIP:DIS: Updating services using commissioning mode 0
CHIP:DIS: Failed to finalize service update: 1c
CHIP:DIS: Delaying proxy of operational discovery: missing delegate
CHIP:IN: CASE Server enabling CASE session setups
CHIP:SVR: Server Listening...
CHIP:DL: Device Configuration:
CHIP:DL:   Serial Number: (not set)
CHIP:DL:   Vendor Id: 65521 (0xFFF1)
CHIP:DL:   Product Id: 32772 (0x8004)
CHIP:DL:   Hardware Version: 0
CHIP:DL:   Setup Pin Code (0 for UNKNOWN/ERROR): 20202021
CHIP:DL:   Setup Discriminator (0xFFFF for UNKNOWN/ERROR): 2560 (0xA00)
CHIP:DL: Invalid manufacturing date: DL
CHIP:DL:   Manufacturing Date: (not set)
CHIP:DL:   Device Type: 257 (0x101)
CHIP:SVR: SetupQRCode: [MT:4CT9142C00C60648G00]
CHIP:SVR: Copy/paste the below URL in a browser to see the QR Code:
CHIP:SVR: https://project-chip.github.io/connectedhomeip/qrcode.html?data=MT:4CT9142C00C60648G00
CHIP:SVR: Manual pairing code: [23331712339]
CHIP:DIS: Updating services using commissioning mode 1
CHIP:DIS: Advertise commission parameter vendorID=65521 productID=32772 discriminator=2560/10 cm=1
CHIP:DIS: Failed to advertise commissionable node: 3
CHIP:DIS: Failed to finalize service update: 1c
[0m[M4 APPLICATION] BLE advertising started. Waiting for Pairing.
[0m[M4 APPLICATION] App Task started
CHIP:DL: kAsyncInitCompleted done
CHIP:DL: CHIPoBLE start advertising
CHIP:DL: state return update adv: 0
First index in 0 state 

[RTSM] - BLE LL FSM 0 -> 1
[RTSM] - BLE LL FSM 0 -> 1
[RTSM] - BLE APPLICATION STARTS ADV => Radio granted to BLEcheck set discoverable , result: 0 
Successfully Start Fast Advertising 

[RTSM] - BLE LL FSM 1 -> 2EVT_LE_CONN_COMPLETE for connection handle 0x801
CHIP:DL: Gap connect
CHIP:DL: BLE GATT connection established 
CHIP:DL: CHIPoBLE advertising stopped
[... 809 lines cut ...]
OK|xfa,0000]	600c -> 6000	(17)

(0032096 > 7896)	[TXa:xfb|00010000|..|xfb,0000]	600c -> 6000	(17)

CHIP:IN: (U) Sending msg 72792689 to IP address 'UDP:[fdae:651e:36d7:3a74:9812:753f:b074:e094%ot]:36487'
(0053792 > 7896)	[TXa:xfc|OK|xfc,0000]	600c -> 6000	(17)

			[RXa:x15|B0(0/1)|IT:172|192,0]	6000 -> 600c	(17)

CHIP:EM: >>> [E:31507r M:102094466 (Ack:72792689)] (U) Msg RX from 0:0128CFBC1F76FD3F [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
			[RX :x28|B0(0/1)|IT:173|192,0]	0100 -> 6000	(17)

[RTSM] - 000002b2|0|0			[RX :x29|B0(0/1)|IT:174|192,0]	0100 -> 6000	(17)

			[RX :x78|B0(0/1)|IT:175|56,0]	cadd -> ffff	(17)

			[RX :x2b|B0(0/1)|IT:176|192,0]	5621 -> 6000	(17)

			[RX :x2c|B0(0/1)|IT:177|189,0]	5621 -> 6000	(17)

			[RX :x2d|B0(0/1)|IT:178|192,0]	5621 -> 6000	(17)

			[RX :x2e|B0(0/1)|IT:179|192,0]	5621 -> 6000	(17)

			[RX :xd1|B0(0/1)|IT:180|161,0]	2505 -> ffff	(17)

STYoannZamaron, 2023-07-07:
Hello can you try again with this log at 1  #define CHIP_DETAIL_LOGGING 1 in CHIPProjectConfig.h
Thanks
Yoann

Gonta01, 2023-07-10:
In one cases device doesn't connect to Google Home. In another it could connect but without any controls available.
This time device connected to Google Home but it isn't possible to control it.
 #define CHIP_DETAIL_LOGGING 1 was set in CHIPProjectConfig.h
Logs: [stm32wb_matter_log_2.log](https://github.com/stm32-hotspot/stm32wb-matter-device-over-thread/files/11998831/stm32wb_matter_log_2.log)

STYoannZamaron, 2023-07-10:
Hello,
When you are doing the commissioning, the controller is losing the ble link with the end devices before the end of the commissioning. That why you cannot control the end device after because for him you didn t receive the commissioning complete from the controller.

Gonta01, 2023-07-10:
Hello. 
Could you please explain what logs have given you such information?
What are possible reasons of this problem and how to fix them?
Also I see in the logs the following message: `CHIP:DMG: CHIP_ERROR_PERSISTED_STORAGE_NOT_FOUND`
Could it be the problem also?

Gonta01, 2023-07-12:
In CHIPProjectConfig.h file there is such lines of code:
```
/**
 * CHIP_DEVICE_CONFIG_DEVICE_PRODUCT_ID
 *
 * 0x8005: example lighting app
 */
#define CHIP_DEVICE_CONFIG_DEVICE_PRODUCT_ID 0x8004
```
Does it mean that for Lighting app example I need to change Product Id on 0x8005?
But as I see the QR code is for 0x8004 product ID.

STYoannZamaron, 2023-07-12:
hello, this software is compatible with matter chip-tool controller V1.0.0 and will not be compatible with other versions of chip-tool. Product_id can stay 0x8004 for a lighting application.

Gonta01, 2023-07-12:
hello, thank you for your answers and explanations. Continue to investigate the problem and I have an assumption. Maybe the cause of the problem is that BLE and Thread for some reason can't work simultaneously on microcontroller?
Radio granted to 15.4.  15.4 stands for Thread as I understand.
```
[RTSM] - BLE LL FSM 2 -> 0
[RTSM] - BLE LL FSM 2 -> 0
[RTSM] - BLE DISCONNECTION Reason : BLE_REASON_MASTER_CLOSE => Task to switch Radio
[RTSM] - TASK Trigged Switch => Radio granted to 15.4
```

Gonta01, 2023-07-12:
The strange thing is that in some cases commissioning is successful. But still it isn't possible to control device in app.
For example logs: 
[stm32wb_matter_log_5.txt](https://github.com/stm32-hotspot/stm32wb-matter-device-over-thread/files/12028563/stm32wb_matter_log_5.txt)

STYoannZamaron, 2023-07-13:
hello can you try with chip-tool :  https://github.com/project-chip/connectedhomeip/tree/v1.0-branch  with RPi + Boarder Router.
BR

Gonta01, 2023-07-14:
Are these logs generated by wireless coprocessor?
```
[RTSM] - BLE LL FSM 2 -> 0
[RTSM] - BLE LL FSM 2 -> 0
[RTSM] - BLE DISCONNECTION Reason : BLE_REASON_MASTER_CLOSE => Task to switch Radio
[RTSM] - TASK Trigged Switch => Radio granted to 15.4
```
 There isn't any functions that generate such logs in source files. Is this implemented in `stm32wb5x_BLE_Thread_ForMatter_fw.bin`?

STYoannZamaron, 2023-07-14:
hello, yes those logs are generated by the RTSM module. this module provides the switch between the ble ip et thread ip.

Gonta01, 2023-07-14:
Get it. Is it possible to figure out the reason why BLE disconnects before commissioning finished in the case of Nest hub? Or it is hidden in binary files so the reason cannot be figured out from my side?

STYoannZamaron, 2023-07-15:
if it s only with google. the best way would be to retrieve the google home log with adb. I think the google home app close the ble link with the end device.

closed 2023-09-01 by STYoannZamaron