stm32-hotspot/stm32wb-matter-device-over-thread #8: Longer operation or continuous sending of commands will result in loss of response

NNAA0917, 2023-08-01:
### Environment　
- NUCLOE-WB55RG
- Home Pod mini
- iPhone SE
### Reproduction steps
1. Clone stm32wb-matter-device-over-thread Tags v1.0.0.
2. Build the cloned git using STM32CubuIDE.
3. Write the following binaries using STMcubuProgrammer (written binaries and locations are listed below).
-  stm32wb5x_BLE_Thread_ForMatter_fw.bin [0x0807F000] because FUS Version is v1.2.0.0
-  Lighting-App.bin or Window-App.bin [0x08000000]
4. Perform commissioning using the iPhone Home app and successful commissioning and control of the device.
5. Prolonged use or continuous sending of commands (though GUI controlled on the home app) will result in loss of response.
When the debugging function of CubeIDE was used to check the situation at the time of response, the following function is stuck in an infinite loop.
[**static void Wait_Getting_Ack_From_M0(void)**]
It appears that FlagReceiveAckFromM0 is not getting out of the loop because it is not getting up.Below is the status of the Task at that time

| Taskname | Status |
| :---: | :---: |
| APPTASK  | Blocked |
| IDLE | Ready |
| PUSH_BUTTON_EVT_PROCESS| Suspended |
| SHCI_USER_EVT_PROCESS| Blocked |
| STM32WB TASK | Running | 
| THREAD_MSG_M0_TO_M4_PROSESS| Suspended |
| Tmr Svc | Blocked |

### Bug prevalence
Always
###

STYoannZamaron, 2023-08-01:
Hello,
can you try with the latest version (v1.1.0). We fix some deadlock between M4 and M0.
BR
Yoann

NNAA0917, 2023-08-02:
Thank you for your response.
I will give it a try.
If you don't mind, could you please let us know what fixes you made to resolve the issue?

NNAA0917, 2023-08-07:
We were able to confirm that the response was successful using v1.1.0. Thank you very much.

closed 2023-08-07 by STYoannZamaron