stm32-hotspot/stm32wb-matter-device-over-thread #9: Cannot reconnect when power is turned off after commissioning

NNAA0917, 2023-08-01:
### Environment　
- NUCLOE-WB55RG
- Home Pod mini
- iPhone SE
### Reproduction steps
1. Clone stm32wb-matter-device-over-thread Tags v1.0.0.
2. Build the cloned git usin
[ConnectionErrorWhenRepowering.log](https://github.com/stm32-hotspot/stm32wb-matter-device-over-thread/files/12223900/ConnectionErrorWhenRepowering.log)
g STM32CubuIDE.
3. Write the following binaries using STMcubuProgrammer (written binaries and locations are listed below).
-  stm32wb5x_BLE_Thread_ForMatter_fw.bin [0x0807F000] because FUS Version is v1.2.0.0
-  Lighting-App.bin or Window-App.bin [0x08000000]
4. Perform commissioning using the iPhone Home app and successful commissioning and control of the device.
5. The device cannot rejoin the Thread Network when the device is turned off or the reset button is pressed and the device is turned on again.
The log on the serial at this time is attached.
### Bug prevalence
Always
###

STYoannZamaron, 2023-08-01:
Hello,
Thank you for the feedback. We are aware of this issue and working as fast as possible to provide a patch.
Workaround: Reboot border router (homepod mini) and after reboot end device. This will lead to flush the srp in the border router and the end device will create a new one.
br
Yoann

NNAA0917, 2023-08-02:
Thank you for your response. We look forward to hearing from you.

STYoannZamaron, 2023-09-01:
hello we did a patch to resolve this issue.

NNAA0917, 2023-09-05:
Thank you for your response.
Do you mean that it has already been supported in v1.1.1?
I have tried with v1.1.1, but the connection does not continue after commissioning, powering down, and powering back up.
The implementation is as follows.

1. build v1.1.1 with STM32CubeIDE and start debugging
2. perform commissioning via Home Pod in Apple Home app.
3. commissioning is completed
After a few moments of operation, press SW1 and [SAVE NVM] will appear in the serial log
5. remove the power and turn it on again after a while 7.
6. check for device response, but there is no response

Is there a problem with the procedure?

STYoannZamaron, 2023-09-05:
Hello no the procedure is ok. I think you face another issue who block the join from the child (end device) to the border router. We are working on this one. Can you try to do 1 commissioning (after commissioning the nvm will be save automatically), some on/off and reboot your device (hard reboot) and if you face the error can you share with us the log of the end device.

NNAA0917, 2023-09-05:
Hello.
Thanks for the confirmation.
I see that the ability to reconnect when power is turned off and on again after commissioning is completed has not yet been implemented.
We will provide a log of the failure at that time.
The [fail_ResetButton.lo
[fail_PowerOff.log](https://github.com/stm32-hotspot/stm32wb-matter-device-over-thread/files/12523083/fail_PowerOff.log)
g] is the log when the reset button is pressed.
[fail_PowerOff.log] is the log when the power is disconnected.
One question, then, what was fixed from v1.1.0 to v1.1.1?
[fail_ResetButton.log](https://github.com/stm32-hotspot/stm32wb-matter-device-over-thread/files/12523032/fail_ResetButton.log)

NNAA0917, 2023-09-06:
Thank you for your response.
To recap.
1. the implementation of the restart regarding the Matter protocol is complete.
2. due to thread communication issues, we are unable to reconnect to the thread border router on restart.
Is that correct?
I sometimes get no response after commissioning or after commissioning is completed (Task is deadlocked), can you confirm this?

STYoannZamaron, 2023-09-06:
hello
1) yes
2) yes but the error is random and not occur all the time.
Can you share the callstack when you see a deadlock.
best regards

NNAA0917, 2023-10-31:
Thank you for your response.
I have tried v1.1.1.
Commissioning is stable and operational, but after commissioning and removing power, the connection does not resume and becomes unresponsive. Is this expected behavior?

Also, the following are listed as changes in v1.1.1, can you tell us what exactly has been improved?

1.Slow Bluetooth advertising added.

2.Factory Data provider implementation added.

3.SRP credentials saved after a power off.

4.NVM size optimised.

Thank you in advance.

NNAA0917, 2023-11-07:
I will add a note.
In v1.1.1, we have confirmed that when power is reconnected after commissioning, the response is properly restored.
One point, however, we have confirmed that if the SSID of the connected Thread Border Router is changed, the response is lost.
Why is this? Thank you in advance.